autowp/arduino-mcp2515 #3: How to define a Extended Frame Format ?

HaganHao, 2017-10-29:
I have try to send a message within Extended Frame Format.
When I use a USB-CAN Analyst to read the message, it show me wrong ID, And I was told it is Stranded Frame Format.

autowp, 2017-10-29:
Add [CAN_EFF_FLAG](https://github.com/autowp/arduino-mcp2515/blob/master/can.h#L13) to message id

```
struct can_frame canMsg;

canMsg.can_id  = 0x0F6 | CAN_EFF_FLAG;
canMsg.can_dlc = 1;
canMsg.data[0] = 0x8E;
```

HaganHao, 2017-10-31:
Thank you very much. works good.

autowp, 2017-10-31:
Welcome

closed 2017-10-31 by autowp

sebasira, 2019-04-03:
Hi! Thanks for your work I appreciate that. This issue was very useful to me because the README is worng, it says **CAN_EFF_MASK;** instead of **CAN_EFF_FLAG*. And it was not working until I use the corerct name.

Please correct for future users!

autowp, 2019-04-03:
@sebasira thanks, fixed

ShrishgovindTork, 2023-03-22:
@autowp what can I do to receive Extended CAN Frame??