autowp/arduino-mcp2515 #6: Weird message behavior

GeorgeTG, 2017-11-07:
I am experiencing some weird message behavior.
I am using your library since i like its API and the ability to run 20 MHZ crystals on the mcp2515 which is what I am using on my custom boards.
On the bus I have 4 cm of wire terminated on both boards with 120R resistors. The boards have common grounds.
Running the example can_write and can_read on each board respectively, I am getting only one of the two messages on the receiver side, and some times, especially when i reset the sender, I get the messages but as seen in the screenshot. Any idea why this might happen? Thanks in advance.
[Imgur](https://i.imgur.com/osJ26ns.png)

autowp, 2017-11-07:
Try to add a pause between two sendMessage. Maybe library trying to put frame in same output buffer

GeorgeTG, 2017-11-08:
Actually I tried to change my crystal to 16MHZ (and of course the change to MCP_16MHZ) and everything worked fine. I am gonna test with another board at 20MHZ to confirm this

autowp, 2017-11-08:
If you found valid configuration please add numbers to library https://github.com/autowp/arduino-mcp2515/blob/master/mcp2515.h#L124

closed 2025-02-26 by autowp