autowp/arduino-mcp2515 #7: 83,333

mcGradus, 2018-05-01:
Здравствуйте, у меня ничего не принимает и не отправляет в мерседес w220 по 83.333... Дело точно не в подключении, с библиотекой mcp_can.h все прекрасно работает, а вот ваша библиотека молчит, то есть если одно устройство с mcp_can.h будет отправлять то другое устройство с вашей библиотекой на 83,333 не услышет ничего а на 250 и выше все слышит и передает... В машине соответственно работает только mcp_can.h ... Подскажите пожалуйста, как можно их подружить? Очень хочется канхакер, но ему если принудительно в canhacker.cpp задать скорость 83.333, то в эфире тоже тишина...

autowp, 2018-05-03:
Сравните тайминги для 83.3 в работающий и нашей библиотеках.

mcGradus, 2018-05-03:
Если
#define MCP_16MHz_83k3BPS_CFG1 (0x03)
#define MCP_16MHz_83k3BPS_CFG2 (0xBE)
#define MCP_16MHz_83k3BPS_CFG3 (0x07)
это оно, то это совпадает, если где в другом месте, то подскажите пожалуйста... Да, и еще на этой скорости в принципе RX TX диоды не загораются на шилде, если просто включить передачу сообщения, а на скорости 500к загораются... Боюсь, не в таймингах дело...

autowp, 2018-05-03:
А проблема только в паре с канхакером (либой arduino-canhacker) или сама по себе наша библиотека mcp2515 тоже не срабатывает?

mcGradus, 2018-05-03:
Именно текущую. Пока пытаюсь заставить работать только скетч  can_read с именно этой библиотекой на 83.333. Отставить канхакер. Про него написал только чтобы объяснить почему борись именно за вашу библиотеку а не использую другую.

mcGradus, 2018-05-03:
Да два устройства с текущей библиотекой общаются нормально, но не чего не слышат в машине

mcGradus, 2018-05-03:
Я короче щас сниму видео и выложу

autowp, 2018-05-03:
Одна и таже ардуина с одной и той же машиной на одной и той же скорости 83.333 работает (mcp_can) и не работает (arduino-mcp2515), верно?

mcGradus, 2018-05-03:
Верно. Именно на 83.333. На 500 работает

mcGradus, 2018-05-03:
Видео снял. Гружу на ютуб.

mcGradus, 2018-05-03:
https://www.youtube.com/watch?v=bQ257H0LkgM

autowp, 2018-05-03:
Посмотрите, какие интерфейсные микросхемы используются в исследуемом автомобиле.
Предположу, что там, где 83.333 используется FT-CAN (Fault Tolerant). Он несовместим с обычным CAN (хотя иногда может по случайности работать).

Если это так, то смотрите в сторону микросхем TJA1055T и её аналогов

mcGradus, 2018-05-03:
а Вы уже видео посмотрели? ведь с mcp_can в машине то же самое железо все прекрасно принимает...

autowp, 2018-05-03:
Бывает, что работает. Когда я сам впервые наступил на эти грабли (если мой предположение верно), я тоже кратковременно добился работы с MCP2551 варьируя длину проводника. 

Впринципе, добиться работы FT-CAN драйвера в сети CAN можно, но это не инженерный подход - это про удачу.

Было даже такое: в FT-CAN машине стоял эмулятор ченджера с "неправильным" CAN и он успешно выполнял свою работу. В тоже время подключенный к шине FT-CAN canhacker показывал только сообщения от самой машины, игнорируя таковые от CAN ченджера.

Я бы предостерег от использования неподходящего CAN не только потому, что это некрасиво и требует удачи, но ещё потому, что это может растить счетчики ошибок в устройствах, подключенных к шине

mcGradus, 2018-05-03:
В другом ардуино на такой же микросхеме FT-СAN? Просто пусть заработает передача из ардуино в "ардуино с точно тем же железом и любой другой кан библиотекой" на скорости 83,333 и я бы успокоился... А пока бессонные ночи.. Забегая вперед, этих канов куплено "ведро", и в тачке они работают, просто процесс изыскания встал когда вернул знакомому его канхакер новосибирский, и стал искать ему замену... Походу придется все таки покупать новосибирский для отлова данных и собирать готовое с другой библиотекой... Извините если напряг. Ну или второй вариант: могу быть супербагфайндером и допилим Ваш Суперпродукт. По моему вы видео не смотрели.

autowp, 2018-05-03:
Зайдите пожалуйста в telegram и напишите мне на @autowp - разберемся. Есть видение

autowp, 2018-05-03:
Resolved in https://github.com/autowp/arduino-mcp2515/compare/16792fd6e8b8...68c80ecb0df2

closed 2018-05-03 by autowp