autowp/arduino-mcp2515 #15: Problem with Arduino Due

4ayk0, 2019-04-21:
I`m trying to use your library with Arduino Due and it won`t work.
I spent some time to make it working and I found solution.
Two line were commented in the file mcp2515.cpp

MCP2515::MCP2515(const uint8_t _CS)
{
    //SPI.begin();  //Commented

    SPICS = _CS;
    pinMode(SPICS, OUTPUT);
    //endSPI(); //Commented
}
After this change Due successfully works with your samples and Can Hacker works too.
Are you sure SPI.begin() is needed here? All your examples also contains SPI.begin() in Setup.

AlexandreBras, 2019-04-24:
Hello, 

I think I have the same problem. Did you had to comment the SPI.begin() and endSPI() lines in order to work?

Does it work fine with this lines commented?

AlexandreBras, 2019-04-24:
Never mind.

It is working fine with those lines commented.

autowp, 2019-04-24:
You may add solution to repo via macro or "isArduinoDue" flag in object

4ayk0, 2019-04-24:
Arduino Uno also works with those lines commented.

milesfrankland, 2019-12-09:
On an arduino zero it doesnt boot up the board unless you comment SPI.begin().
You can leave the endSPI(); though

zentT, 2021-01-05:
I canno't make this work with an Arduino Due and a MCP2515, even if i comment the lines you mentioned. After   mcp2515.reset(); its blocked. Any advice would be greatly appreciated,

Térence

Lorevalles, 2023-10-28:
Tengo el mismo problema, el programa me funciona correctamente, en un Arduino Every, en el Due se compila sin errores, pero se queda colgado.
Con éste código funciona.
#include <SPI.h>
#include <mcp_can.h>

const int SPI_CS_PIN = 10; // Define el pin CS del MCP2515
MCP_CAN CAN(SPI_CS_PIN); // Inicializa la instancia de la biblioteca MCP_CAN

void setup() {
  Serial.begin(115200);
  // Inicializa la comunicación SPI
  if (CAN.begin(MCP_STDEXT, CAN_125KBPS, MCP_16MHZ) == CAN_OK) {
    Serial.println("MCP2515 Initialized Successfully!");
  } else {
    Serial.println("Error Initializing MCP2515...");
  }
  // Configura la máscara y los filtros CAN si es necesario
  // CAN.init_Mask(0, 0, 0x7FF); // Máscara predeterminada
  // CAN.init_Filt(0, 0, 0x123);  // Filtro predeterminado
  // CAN.init_Filt(1, 0, 0x456);  // Filtro adicional
  // ...
}

void loop() {
  // Tu código para enviar y recibir mensajes CAN
  // Ejemplo de envío de un mensaje CAN:
  // unsigned char msg[8] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08};
  // CAN.sendMsgBuf(0x123, 0, 8, msg);
  
  // Ejemplo de recepción de un mensaje CAN:
  // if (CAN.checkReceive()) {
  //   unsigned char len = 0;
  //   unsigned char buf[8];
  //   if (CAN.readMsgBuf(&len, buf) == CAN_OK) {
  //     Serial.print("Received a CAN message: ");
  //     for (int i = 0; i < len; i++) {
  //       Serial.print(buf[i], HEX);
  //       Serial.print(' ');
  //     }
  //     Serial.println();
  //   }
  // }
}

Lorevalles, 2023-10-28:
Conexión:

Conecta los pines del MCP2515 a los pines del Arduino Due de la siguiente manera:

MCP2515 VCC a 5V del Arduino Due.
MCP2515 GND a GND del Arduino Due.
MCP2515 SCK a pin SPI SCK (pin 76).
MCP2515 SO a pin SPI MISO (pin 74).
MCP2515 SI a pin SPI MOSI (pin 75).
MCP2515 CS a un pin de tu elección (por ejemplo, pin 10) y asegúrate de que coincida con la configuración en tu código.
MCP2515 INT a un pin de tu elección que soporte interrupciones (por ejemplo, pin 2) y asegúrate de que coincida con la configuración en tu código.

Lorevalles, 2023-10-28:
El Arduino Due tiene 54 pines digitales y 12 pines analógicos disponibles. Algunos de estos pines están reservados para funciones específicas, como UART, SPI y PWM, por lo que debes elegir pines disponibles para conectar el MCP2515. Además, el Arduino Due tiene soporte para interrupciones en varios de sus pines digitales y analógicos, lo que podría ser útil si deseas utilizar la función de interrupción del MCP2515.

La elección de los pines dependerá de tu proyecto y de los pines disponibles en tu placa. Aquí hay algunas sugerencias:

SPI (SCK, MISO, MOSI): Estos pines son fijos en el Arduino Due y están ubicados en los pines 75 (SCK), 74 (MISO) y 76 (MOSI).

Chip Select (CS): Elije un pin digital disponible para el CS del MCP2515, como el pin 10, 52 o cualquier otro pin disponible en tu placa.

Interrupción (INT): Utiliza un pin que admita interrupciones. Por ejemplo, el pin digital 2 o 3 se pueden utilizar para la interrupción del MCP2515. Asegúrate de configurar correctamente la interrupción en tu código.

Asegúrate de modificar el código de configuración para que coincida con los pines específicos que hayas elegido para conectar el MCP2515. Además, ten en cuenta cualquier cambio necesario en la configuración de máscaras y filtros CAN según las necesidades de tu aplicación.

Lorevalles, 2023-10-28:
Lo he implementado en éste programa. He dejado comentado la versión que funciona en un Arduino Every.
# Programa

Lorevalles, 2023-10-28:
/*
  Este código es una implementación para Arduino Nano Every
  utilizando una pantalla SSD1309 y un módulo CAN
  (Controller Area Network). El código controla y muestra información sobre el
  estado de la batería de alto voltaje sustituida en un Toyota Prius Gen 2.
  Realiza comunicación a través del bus CAN. Consiguiendo su optimización de funcionamiento.
  https://attachments.priuschat.com/attachment-files/2021/09/211662_Prius22009_CAnCodes.pdf
  https://en.wikipedia.org/wiki/OBD-II_PIDs#Standard_PIDs
  https://es.wikipedia.org/wiki/OBD-II_PID
  https://www.eaa-phev.org/wiki/Prius_PHEV_TechInfo#4
  https://github.com/Lorevalles/Proyecto_Prius
*/
#include <U8g2lib.h>
#include <SPI.h>
#include <mcp_can.h>
// #include <mcp2515.h>

const int SPI_CS_PIN = 10;  // Define el pin CS del MCP2515
const int INT_PIN = 2;
const int SS_RX_PIN = 11;
const int SS_TX_PIN = 12;

U8G2_SSD1309_128X64_NONAME0_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/5, /* data=*/6, /* cs=*/7, /* dc=*/9, /* reset=*/8);
//struct can_frame canMsg;
//MCP2515 mcp2515(SPI_CS_PIN);

MCP_CAN CAN(SPI_CS_PIN);  // Reemplaza SPI_CS_PIN con el número de pin que estás utilizando como chip select

///unsigned long lastAvisoRvTime = 0;
const unsigned long avisoRvInterval = 1000;  // Intervalo de tiempo en milisegundos (1 segundo)
unsigned long lastAvisoRcTime = 0;
const unsigned long avisoRcInterval = 1000;  // Intervalo de tiempo en milisegundos (1 segundo)

unsigned long currentMillisEv = millis();  // Obtener el tiempo actual
unsigned long ultimaActualizacion = 0;     // Variable para almacenar el tiempo de la última actualización
int valorAnterior = -1;                    // Inicialmente, ningún valor anterior

// Declarar una variable para almacenar el tiempo del último cambio en el valor de lecturas por segundo (L:)
unsigned long lastReadingsUpdateMillis = 0;
// Declarar una variable para almacenar el tiempo del último cambio en el valor de mensajes enviados por segundo (E:)
unsigned long lastSentMessagesUpdateMillis = 0;

// Variables globales Cotrol activaEv
bool activacionPendiente = false;
unsigned long ultimaActivacion = 0;
unsigned long tiempoUltimaAplicacion = 0;
unsigned long tiempoSinDatos = 0;
const unsigned int evTiactivado = 300;     // 300 milisegundos Tiempo pulsado el boton
const unsigned int tiempoEsperar = 30000;  // 30 segundos en milisegundos Ciclo de verificación

// Inicio encendido control
int tiempoEspera = 10000;  // Espera para poder apagar el vehiculo al reiniciar

// Volores control A0 A1 y D3, D4
const byte socBajo = 70;         //75 Valor a mantener minimo de SOC
const byte socAlto = 75;         //77 Valor a mantener Maximo de SOC
const byte limiteCarga = 80;     //80 Comienza a descargar
const byte voltajeMinimo = 228;  // 215
const byte voltajeMaximo = 238;  // 230
const byte marcaBaja = 65;       // Este valor tambien condiciona a partir de carga voltajeMaximo
const byte marcaAlta = 80;       // Marca en icono SOC

// Variables para el control de la salida A0 (Rele Voltaje)
unsigned long lastOutputA0ChangeMillis = 0;   // Último cambio de estado
const unsigned int tiempoActivoA0 = 1000;     // 0.5 segundos en milisegundos
const unsigned int tiempoInactivoA0 = 30000;  // 30 segundos en milisegundos

// Refresco de pantalla
const unsigned int esperaRef = 500;  // 1000 cada segundo actualiza los datos

[... 1054 lines cut ...]
    // Verifica que al menos uno de los pines D3 o D4 esté alto y cambia su estado a bajo
    if (digitalRead(outputPinD3) == HIGH || digitalRead(outputPinD4) == HIGH) {
      d3Activated = false;
      lastHvSocValue = socAlto + 1;
      // Cambia los pines D3 y D4 a bajo
      digitalWrite(outputPinD3, LOW);
      // digitalWrite(outputPinD4, LOW);
    }
  }

  // Actualiza el estado anterior del botón y el interruptor
  lastButtonState = buttonState;
  lastSwitchState = switchState;

  /////////////////////////////////////////////////////////////////////
  /* if (lastHvSocValue <= socBajo) {
    digitalWrite(outputPinD3, HIGH);
    d3Activated = true;
  }*/
  leerCan();
  activaEv();
  conCarga();
  lecSegundo();
  controlSoc();
}