autowp/arduino-mcp2515 #21: Mcp2515 does not send in normal mode

mvturnho, 2019-07-18:
I am running this driver on esp8266 devices. When I have the mcp2515 in loopbackmode I can send and receive data. This indicates that my spi connections are ok. When I use the example read and write programs on two separate esp8266 with mcp2515. The sender fills up the 3 message queues in the mcp2515 but the receiver shows no data.
Is this driver suited for esp devices? And what can I check to see what is wrong?

autowp, 2019-07-18:
Yes, we tested esp config: it works.

mvturnho, 2019-07-19:
Ok so I am doing something wrong. 
I have the 8Mhz devices from Aliexpress. first tried:
```
  mcp2515.reset();
  mcp2515.setBitrate(CAN_125KBPS, MCP_8MHZ);
  mcp2515.setNormalMode();
```
Then swapped out the 8Mhz for 16Mhz changed code to   `mcp2515.setBitrate(CAN_125KBPS, MCP_16MHZ);` without result.

I made the folowing connections:
```
esp8266  MCP2515
14 = D5      sck
12 = D6      so => miso
13 = D7      si => mosi
15 = D8      cs
```
added the jumpers for bus termination on both devices.
connected both mcp2515 devices with `CAN_H`, `CAN_L `and `gnd`. (`CAN_H `to `CAN_H `and `CAN_L `to `CAN_L`)

when I use the folowing code whe I use the loopback it works, so I think my spi connections are OK.

```cpp
#include <SPI.h>
#include <mcp2515.h>

#define MCP_CS 15

struct can_frame canMsg1;
struct can_frame canMsg;

MCP2515 mcp2515(MCP_CS);

void setup() {
  canMsg1.can_id = 0x0F6;
  canMsg1.can_dlc = 8;
  canMsg1.data[0] = 0x8E;
  canMsg1.data[1] = 0x87;
  canMsg1.data[2] = 0x32;
  canMsg1.data[3] = 0xFA;
  canMsg1.data[4] = 0x26;
  canMsg1.data[5] = 0x8E;
  canMsg1.data[6] = 0xBE;
  canMsg1.data[7] = 0x86;

  Serial.begin(115200);
  SPI.pins(PIN_SPI_SCK,PIN_SPI_MISO, PIN_SPI_MOSI,PIN_SPI_SS);
  SPI.begin();

  mcp2515.reset();
  mcp2515.setBitrate(CAN_125KBPS, MCP_16MHZ);
  mcp2515.setLoopbackMode();
  delay(500);
  Serial.println("\n\nExample: Write to CAN");
}

void loop() {
  mcp2515.sendMessage(&canMsg1);

  // for loopback check
  if (mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK) {
    Serial.print(canMsg.can_id, HEX);  // print ID
    Serial.print(" ");
    Serial.print(canMsg.can_dlc, HEX);  // print DLC
    Serial.print(" ");
    for (int i = 0; i < canMsg.can_dlc; i++) {  // print the data
      Serial.print(canMsg.data[i], HEX);
      Serial.print(" ");
    }
    Serial.println();
  }

  delay(100);
}
```

Then I set it back to   `mcp2515.setNormalMode();`  and remove the `mcp2515.readMessage(&canMsg) `from the loop.

on the second esp8266 with its own mcp2515 I use a reader;

```cpp
#include <SPI.h>
#include <mcp2515.h>

#define MCP_CS 15

struct can_frame canMsg;
MCP2515 mcp2515(MCP_CS);

void setup() {
  Serial.begin(115200);
  SPI.pins(PIN_SPI_SCK,PIN_SPI_MISO, PIN_SPI_MOSI,PIN_SPI_SS);
  SPI.begin();

  mcp2515.reset();
  mcp2515.setBitrate(CAN_125KBPS, MCP_16MHZ);
  mcp2515.setNormalMode();
  delay(500);

  Serial.println("\n\n------- CAN Read ----------");
  Serial.println("ID  DLC   DATA");
}

void loop() {
  if (mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK) {
    Serial.print(canMsg.can_id, HEX); // print ID
    Serial.print(" ");
    Serial.print(canMsg.can_dlc, HEX); // print DLC
    Serial.print(" ");
    for (int i = 0; i < canMsg.can_dlc; i++) { // print the data
      Serial.print(canMsg.data[i], HEX);
      Serial.print(" ");
    }
    Serial.println();
  }
}
```

Any ideas on what is wrong? I work on [esphome](https://github.com/mvturnho/esphome/tree/canbus) where I want to add canbus communication to the setup. your driver has the cleanest and most functional api, so that is why I keep trying getting this to work.

mvturnho, 2019-07-20:
I think I found the issue that keeps the mcp2515 from sending/receiving.

My mcp2515 board has a tja1050 transceiver. These are 5v chips. So my 3.3v supply from the esp8266 is insufficient.
I ordered the SN65HVD230 transceivers to replace them.

closed 2019-07-20 by mvturnho