autowp/arduino-mcp2515 #24: Extended frame returning wrong MSB ID value

OiD-W, 2019-08-12:
I'm filtering EXT ID's and while the filters are working the returned value is not correct.
Here is the code (from the example receive sketch);
```
#include <SPI.h>
#include <mcp2515.h>

struct can_frame canMsg;
MCP2515 mcp2515(10);

void setup() {
  Serial.begin(115200);
  SPI.begin();
  
  mcp2515.reset();
  mcp2515.setBitrate(CAN_125KBPS, MCP_8MHZ);
  mcp2515.setConfigMode();
  mcp2515.setFilterMask(0,false, 0x1FF);
  mcp2515.setFilter(0, false, 0x1BA);
  mcp2515.setFilter(1, false, 0x1BA);
  mcp2515.setFilterMask(1,true,0x1FFFFFFF);   //Set mask to compare all bits in EXT ID
  mcp2515.setFilter(2, true, 0x0E345678);     //Set filter for ID 0E345678, returns 8E345678.
  mcp2515.setFilter(3, true, 0x00000001);     //Set filter for ID 00000001, returns 80000001.
  mcp2515.setFilter(4, true, 0x12345678);     //Set filter for ID 12345678, returns 92345678.
  mcp2515.setFilter(5, true, 0x1F00000F);     //Set filter for ID 1F00000F, returns 9F00000F.
  mcp2515.setNormalMode();
  
  Serial.println("------- CAN Read ----------");
  Serial.println("ID  DLC   DATA");
}

void loop() {
  
  if (mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK) {
      
    Serial.print(canMsg.can_id, HEX); // print ID
    Serial.print(" "); 
    Serial.print(canMsg.can_dlc, HEX); // print DLC
    Serial.print(" ");
    
    for (int i = 0; i<canMsg.can_dlc; i++)  {  // print the data
        
      Serial.print(canMsg.data[i],HEX);
      Serial.print(" ");

    }

    Serial.println();      
  }

}
```
The following is giving me the wrong results;

>   mcp2515.setFilter(2, true, 0x0E345678);     //Set filter for ID 0E345678, returns 8E345678.
>   mcp2515.setFilter(3, true, 0x00000001);     //Set filter for ID 00000001, returns 80000001.
>   mcp2515.setFilter(4, true, 0x12345678);     //Set filter for ID 12345678, returns 92345678.
>   mcp2515.setFilter(5, true, 0x1F00000F);     //Set filter for ID 1F00000F, returns 9F00000F.
 
The most significant nibble is returning the ID + 8.

Is this a function? Is this documented?

Thanks and regards,

autowp, 2019-08-13:
Check docs, can.h please.

Most significant bit:
#define CAN_EFF_FLAG 0x80000000UL /* EFF/SFF is set in the MSB */

Always 1 for 29-bit messages

OiD-W, 2019-08-15:
Hi autowp,

Thanks for the quick reply, lots of info in can.h! Might be useful to add some more information to the description field "first commit" like flags and config info.

Thanks for sharing the library!

closed 2019-08-15 by OiD-W

celturbo, 2021-12-30:
@OiD-W 

Managed to solve?

samc1213, 2022-01-18:
@autowp Would it be beneficial to abstract the setting / unsetting of CAN_EFF_FLAG inside the library? This is not expected behavior with CAN in general. While it is documented in can.h, it might help new users by setting / unsetting the flag inside the library so users don't know it is used