autowp/arduino-mcp2515 #36: not entering sleep mode properly

atc1441, 2020-03-20:
Hi, as the SleepMode is not entered most of the times with the current setMode void i suggest the following changes:

put the modifyRegister: https://github.com/autowp/arduino-mcp2515/blob/master/mcp2515.cpp#L167

inside the while loop so it will rewrite the register.

also set the timeout from 10ms to 200ms as the sleep mode entering can take up to 100ms

i also suggest to add the sleepWakeup bit (CANINTF_WAKIF) into the reset void: https://github.com/autowp/arduino-mcp2515/blob/master/mcp2515.cpp#L50

so it will wake up on can messages, or maybe make an extra function to enable or disable that bit so everyone can set if needed.

jxltom, 2020-03-24:
Current code works for me for entering the sleep mode.

Do you mind enlighting me where the specification of that entering sleep mode can takes 100ms? I didn't find it in the datasheet.

atc1441, 2020-03-24:
I tested the time it took to bring it to sleep mode with the edited function, and sometimes it was up to 100ms, it was not from the datasheet.

With the stock function i was not able get it into standby mode by calling it one time, even with disabled timeout.