autowp/arduino-mcp2515 #41: Conflict with CCS811

blazeis, 2020-04-23:
Hello.
I cant init ccs sensor if mcp2515 initialized. Minimal code is:
`#include <SPI.h>
#include <mcp2515.h>
#include "Adafruit_CCS811.h"

Adafruit_CCS811 ccs;
MCP2515 CAN0(10);

void setup() {
  Serial.begin(115200);
  if(!ccs.begin()){
    Serial.println("Failed to start sensor! Please check your wiring.");
    while(1);
  }
}

void loop() {
  delay(5000);
}
`
if i comment out line "MCP2515 CAN0(10);", sensor initializing properly.
Sensor connected to i2c bus. 
Arduino pro mini, atmega 328p 5v 16MHz.

jxltom, 2020-04-23:
MCP2515 is using SPI, CCS811 is using I2C, please check the PIN being used shouldn't have conflicts.

blazeis, 2020-04-24:
Thank you for reply.
At this morning, i check wiring twice. I soldered CS from mcp2515 to pin 10 on the arduino's front side, and wake pin from CCS811 on the arduino's back side.
Sorry  ¯\_(ツ)_/¯

closed 2020-04-24 by blazeis