autowp/arduino-mcp2515 #44: The bit rate macro is not corresponding to the real bit rate.

VoLinhTruc, 2020-06-23:
I set the bit rate is 1Mbps (1us), but the real bit rate is just 500kbps (2us).
I illustrated in the image below.
Please fix this issue.
Thank you.

[image]

oschonrock, 2020-06-28:
I had something similar. The second param to setBitrate is the frequency of the crystal driving your MCP2515 controller. 

On my protyping boards (typical ebay/amazon ones) the crystal is 8Mhz. The param defaults to 16Mhz . 

<mcp2515.h>
```C++
        ERROR setBitrate(const CAN_SPEED canSpeed);
        ERROR setBitrate(const CAN_SPEED canSpeed, const CAN_CLOCK canClock);

... 

MCP2515::ERROR MCP2515::setBitrate(const CAN_SPEED canSpeed)
{
    return setBitrate(canSpeed, MCP_16MHZ);
}
```
So for my board to output the correct frequency bitrate had use the method signature with 2 params and pass in MCP_8MHZ), like this:

```C++
mcp2515.setBitrate(CAN_500KBPS,MCP_8MHZ);
```

Hope that helps...

oschonrock, 2020-06-28:
> PS: I am finding that I get a lot of Error Frames on the Bus at 1Mhz even over just a 10cm Bus with 2 nodes on it. I am yet to find the reason for that.

I think I know the answer. 8Mhz crystals don't support 1Mhz Bus speed, and the 1Mhz config here:
```C++
/*
 *  Speed 8M
 */
#define MCP_8MHz_1000kBPS_CFG1 (0x00)
#define MCP_8MHz_1000kBPS_CFG2 (0x80)
#define MCP_8MHz_1000kBPS_CFG3 (0x80)

#define MCP_8MHz_500kBPS_CFG1 (0x00)
#define MCP_8MHz_500kBPS_CFG2 (0x90)
#define MCP_8MHz_500kBPS_CFG3 (0x82)

....
```

Is at best "marginal -- I measure 950ns bit width which is 5% out when the max tolerance is 1.7% between nodes -- and at worst it is just wrong, because the datasheet says:

```
REGISTER 5-3:
 CNF3: CONFIGURATION REGISTER 3
...

PHSEG2[2:0]: PS2 Length bits
(PHSEG2[2:0] + 1) x TQ. Minimum valid setting for PS2 is 2 TQs.
```
And this lib has it at (0 + 1) = 1 (and that's less than 2) for the 8Mhz crystal + 1Mhz Bus speed case. 

Everything dialled to the max, but it's not quite enough resolution to work reliably, even over ideal electrical Bus conditions.  Timing is too inaccurate. 

TLDR: 1Mhz bus speed is not realiably supporrted with 8Mhz crystals. Need a better prototyping board with a 16 or 20 Mhz Crystal.