autowp/arduino-mcp2515 #47: Adding RTR ID

paulmac00, 2020-09-21:
The library is working great, however I am struggling with adding an RTR tag to the CAN ID. 

Is anyone able to provide a quick pointer? I've been searching the interwebs for the past 3 hours or so 

Thank you!

jxltom, 2020-09-21:
I have a demo here for different kind of message https://github.com/jxltom/micropython-mcp2515/blob/master/demo.py, though it is written in python

paulmac00, 2020-09-21:
Hey JX,

Thanks for the quick response! Any help is very much appreciated. 

I found where you added the RTX tag:
 rtr_frame = CANFrame(can_id=0x7FF | CAN_RTR_FLAG)

I'm working on an arduino so I've written:
frame.can_id = 0x4a | MCP2515::CAN_RTR_BITMASK;

I also look into MCP2515_Const.h where it is defined as
static uint32_t constexpr CAN_RTR_BITMASK   = 0x40000000;

However it doesn't seem to be working properly. I'll play around with it tomorrow and see if I can make some more progress.

jxltom, 2020-09-21:
Can you receive the standard frames? If so, please check the configuration of your RXB. You need to enable it to receive extended frames by ```RXBnCTRL_RXM_STDEXT``` or just disabled the filters/masks.

paulmac00, 2020-09-22:
Hey JX,

**Can you receive the standard frames?**
Yes, I can receive and send standard frames. I will look into the RXB config today.

fkaraokur, 2021-02-09:
I'm having trouble here too. Can you explain what he said?