autowp/arduino-mcp2515 #55: Use of RTR and others

fkaraokur, 2021-02-09:
When I examine the details, the following statement goes!

/ * 32 bit CAN_ID + EFF / RTR / ERR flags * /

So how do we express this in code?

jxltom, 2021-02-10:
Please check this line https://github.com/jxltom/micropython-mcp2515/blob/8d2415f368de271d3b2ba7dea2e7cf9fb2c076cf/demo.py#L57

fkaraokur, 2021-02-10:
> Please check this line https://github.com/jxltom/micropython-mcp2515/blob/8d2415f368de271d3b2ba7dea2e7cf9fb2c076cf/demo.py#L57

I am using mcp2515 with arduino. this is probably the python code.

Actually I don't use RTR. But how should I express this? How to write a statement like this "CAN_no_RTR"?

tx_frame.FIR.B.FF = CAN_frame_ext;
second I need to convert this expression. This is taken from another library.

fkaraokur, 2021-02-10:
```
CAN_frame_t tx_frame;
    tx_frame.FIR.B.FF = CAN_frame_ext;
    tx_frame.MsgID = 0x01100033;
    tx_frame.FIR.B.DLC = 8;
    tx_frame.FIR.B.RTR = CAN_no_RTR;
    tx_frame.data.u8[0] = 0x01;
    tx_frame.data.u8[1] = 0x00;
    tx_frame.data.u8[2] = 0x00;
    tx_frame.data.u8[3] = 0x00;
    tx_frame.data.u8[4] = 0x00;
    tx_frame.data.u8[5] = 0x00;
    tx_frame.data.u8[6] = 0x00;
    tx_frame.data.u8[7] = 0x00;
    ESP32Can.CANWriteFrame(&tx_frame);
```

In summary, I need to convert the above codes. These codes belong to another library. But RTR and EXT are also specified. In this library it is written together. I could not decipher this code.

jxltom, 2021-02-10:
You can use ```prepareId``` method to set the RTR in can ID

fkaraokur, 2021-02-10:
> `prepareId`Kutu kimliğinde RTR'yi ayarlamak için yöntemi kullanabilirsiniz.

example code pls thanks

jxltom, 2021-02-10:
> > `prepareId`Kutu kimliğinde RTR'yi ayarlamak için yöntemi kullanabilirsiniz.
> 
> example code pls thanks

https://github.com/autowp/arduino-mcp2515/blob/cdc0141ee734f47f54628b1ac8acaafa9120511b/mcp2515.cpp#L509

fkaraokur, 2021-02-10:
> > > `prepareId`Kutu kimliğinde RTR'yi ayarlamak için yöntemi kullanabilirsiniz.
> > 
> > 
> > example code pls thanks
> 
> https://github.com/autowp/arduino-mcp2515/blob/cdc0141ee734f47f54628b1ac8acaafa9120511b/mcp2515.cpp#L509

very complicated. And won't the ext and RTR be prepaired at the same time?

jxltom, 2021-02-10:
https://github.com/autowp/arduino-mcp2515/blob/cdc0141ee734f47f54628b1ac8acaafa9120511b/mcp2515.cpp#L590

RTR is just set by specific bit of the can ID