autowp/arduino-mcp2515 #56: How to know if received frame is standard frame or extended frame?

Akshay1595, 2021-02-18:
I want to know if my received frame is the standard or extended frame.

I have the following code to receive a can frame.
`
    while ((mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK))  {
        debug_log("I got some data!");

        Serial.print(canMsg.can_id, HEX); // print ID
        Serial.print(" "); 
        Serial.print(canMsg.can_dlc, HEX); // print DLC
        Serial.print(" ");
        
        for (int i = 0; i < 8; i++)  {  // print the data
            Serial.print(canMsg.data[i],HEX);
            Serial.print(" ");
        }
        Serial.println(" ");
        //debug_log(msg_string);
    }
`

I also want to display the frame format of the above CAN message.
How to do that?

Akshay1595, 2021-02-18:
Hey No worries!

Sorted!
I found this comment line!
`/*
 * Controller Area Network Identifier structure
 *
 * bit 0-28 : CAN identifier (11/29 bit)
 * bit 29   : error message frame flag (0 = data frame, 1 = error message)
 * bit 30   : remote transmission request flag (1 = rtr frame)
 * bit 31   : frame format flag (0 = standard 11 bit, 1 = extended 29 bit)
 */
typedef __u32 canid_t;`

We can check bit 31 and decide if it is Standard or extended frame!

closed 2021-02-18 by Akshay1595

jxltom, 2021-02-19:
Hey, the ```CAN_EFF_MASK``` flag https://github.com/autowp/arduino-mcp2515/blob/cdc0141ee734f47f54628b1ac8acaafa9120511b/can.h#L19 might help.