autowp/arduino-mcp2515 #57: Receive not working

RevDrDR, 2021-03-07:
Hello,

I have two arduinos with a MCP2515 module on each. I installed your library and ran your CAN_write code on one, and your CAN_read code on the other. The sender says it is sending messages and the LED seems to confirm that.

The receiver never gets anything, though. It seems to initialize and prints the header (------- CAN Read ----------
ID  DLC   DATA), but never gets any data. If I put a print where the else of the if would be, it goes there every time. So maybe there is an error? If so, how do I figure out what the error is?

I tried both as senders and both as receivers, and they both seem to send fine, but neither receives. I've also tried with and without terminators and it doesn't seem to matter.

ISO838, 2021-03-29:
@RevDrDR were you able to resolve this? I seem to also have thie problem - just a blank output.

RevDrDR, 2021-03-29:
Nope. We've ordered new CAN modules to see if it might be a hardware issue. But so far, we have not been able to get this code to work.

JackDaviesDesign, 2021-05-06:
Check your frequency setting based on the crystal you have on your board -   mcp2515.setBitrate(CAN_500KBPS, MCP_8MHZ); for 8MHZ crystal

closed 2025-02-26 by autowp