autowp/arduino-mcp2515 #63: can communication between esp32 and arduino not working

StijnVandendriessche1, 2021-04-28:
Hi

I am trying to make an ESP32 (running arduino) communicate with an arduino uno. Both have an MCP2515 board wired up as the following picture shows.

[image]

To enable the can communication via the board I am using the arduino-mcp2515 library. I uploaded the send example to the arduino uno, and the receive example to the ESP32 (changing the chip select to 5 wich is the standard chip select for the SPI bus on the ESP32), yet it is not working.

Does anyone know what I might be doing wrong?

Sending code on the Arduino Uno
`#include <SPI.h>
#include <mcp2515.h>

struct can_frame canMsg1;
struct can_frame canMsg2;
struct can_frame canMsg3;
MCP2515 mcp2515(10);

void setup() {
  canMsg1.can_id  = 0x0F6;
  canMsg1.can_dlc = 8;
  canMsg1.data[0] = 0x8E;
  canMsg1.data[1] = 0x87;
  canMsg1.data[2] = 0x32;
  canMsg1.data[3] = 0xFA;
  canMsg1.data[4] = 0x26;
  canMsg1.data[5] = 0x8E;
  canMsg1.data[6] = 0xBE;
  canMsg1.data[7] = 0x86;

  canMsg2.can_id  = 0x036;
  canMsg2.can_dlc = 8;
  canMsg2.data[0] = 0x0E;
  canMsg2.data[1] = 0x00;
  canMsg2.data[2] = 0x00;
  canMsg2.data[3] = 0x08;
  canMsg2.data[4] = 0x01;
  canMsg2.data[5] = 0x00;
  canMsg2.data[6] = 0x00;
  canMsg2.data[7] = 0xA0;
  
  while (!Serial);
  Serial.begin(115200);
  
  mcp2515.reset();
  mcp2515.setBitrate(CAN_125KBPS);
  mcp2515.setNormalMode();
  
  Serial.println("Example: Write to CAN");
}

void loop() {
  mcp2515.sendMessage(&canMsg1);
  Serial.print("sent ");
  Serial.print(canMsg1.can_id, HEX);
  Serial.println();
  delay(5000);
  mcp2515.sendMessage(&canMsg2);
  Serial.print("sent ");
  Serial.print(canMsg2.can_id, HEX);
  Serial.println();
  delay(5000);
}`

Receiving code on the ESP32
`#include <SPI.h>
#include <mcp2515.h>

struct can_frame canMsg;
MCP2515 mcp2515(5);

void setup() {
  Serial.begin(115200);
  
  mcp2515.reset();
  mcp2515.setBitrate(CAN_125KBPS);
  mcp2515.setNormalMode();
  
  Serial.println("------- CAN Read ----------");
  Serial.println("ID  DLC   DATA");
}

void loop() {
  if (mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK) 
  {
    Serial.print("CAN ID: ");
    Serial.println(canMsg.can_id);
    Serial.print("data: ");
    for (int i = 0; i<canMsg.can_dlc; i++)  {  // print the data
      Serial.print(canMsg.data[i],HEX);
      Serial.print(" ");
    }
    Serial.println();     
  }
}`

Thanks in advance!

closed 2021-07-09 by StijnVandendriessche1

dakhnod, 2021-11-18:
do you have both can modules terminated?
There is a special jumper on the can boards for that.
That's what helped in my case.

StijnVandendriessche1, 2021-11-24:
Hi @dakhnod, thank's for your response. My project has already ended, but I do remember both of the boards being terminated by the on-board resistor. However, I do not remember what solved the problem. Good to hear that you got it working!