autowp/arduino-mcp2515 #69: No data receive

Som4567, 2021-11-12:
Hi

I have made appropriate changes to use 8Mhz crystal but unfortunately, I don't see any data in receiver.

I am not sure what's wrong with my code. Can you please guide me to debug the issue or any ways to figure out the problem.

Just to add: I am using Arduino uno

jacopisky, 2021-11-21:
Hi, it can be possible that your mcp2515 is set in listen only mode. Did you use the examples provided?
Wrong wiring or broken IC may be the problem. Be sure to have properly set the CS pin during the creation of the MCP2515 object according to your board wiring. 
If that does not help, feel free to send me an email.

Som4567, 2021-11-21:
Hi,

Thanks for your response. I don't understand the meaning of MCP2515 is on
listen only mode.How can I change mode of MCP2515 ?

Apart from that I have checked all wiring and IC connection. I am able to
send rpm PID request on CAN.write and able to receive the response but I
couldn't see any message on CAN.receive.

I have checked resistance on car CAN H and CAN L and it was 120 ohm so I
have sorted J1 ( 120 ohm external resister) to make total resistance close
to 60 ohm.

Looking forward to hear from you.

Thanks & Regards.

jacopisky, 2021-11-21:
Ok, if sending goes ok it might be a software issue. The library allows us to set the operating mode of the mcp2515; for example sleep mode or loopback mode and others, please make sure to use the appropriate to your specifications. If you wish, send me or post the code to solve the problem.

jacopisky, 2021-11-21:
Wait, if you are using CAN library, this is not its repository. I'm guessing you are using a virtualization of the MCP2515 to operate with a third part's library

Som4567, 2021-11-22:
Yes, you are right, just to test I have used MCP2515 CAN library from Sandeep mistry and I am able to see the response of OBD II - >RPM example but even with his library I am not able to receive any CAN message without sending CAN.write. 

Today morning I got MCP2515 with MCP2551 along with 16Mhz crystal so need not to make any changes in your example code except 'CAN_500KBPS' as my car communicates at 500KBPS speed. 

Just to add  - In my new CAN - BUS board 120 ohm external resister is there so after connecting to OBD port I am getting 60.3 ohm across CAN H and CAN L.

jacopisky, 2021-11-22:
Sorry, I have to ask you to post the code to be sure to solve the problem

Som4567, 2021-11-22:
`#include <SPI.h>
#include <mcp2515.h>

struct can_frame canMsg;
MCP2515 mcp2515(10);

void setup() {
  Serial.begin(115200);
  
  mcp2515.reset();
  mcp2515.setBitrate(CAN_500KBPS);
  mcp2515.setNormalMode();
  
  Serial.println("------- CAN Read ----------");
  Serial.println("ID  DLC   DATA");
}

void loop() {
  if (mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK) {
    Serial.print(canMsg.can_id, HEX); // print ID
    Serial.print(" "); 
    Serial.print(canMsg.can_dlc, HEX); // print DLC
    Serial.print(" ");
    
    for (int i = 0; i<canMsg.can_dlc; i++)  {  // print the data
      Serial.print(canMsg.data[i],HEX);
      Serial.print(" ");
    }

    Serial.println();      
  }
}`
I am using the above code. 

BTW I purchased [galileosky ](https://galileosky.com/products/devices/obd-ii/galileosky-obd-ii.html) OBD and their support team pointed out that my car is just supporting J1979 protocol ie 'request-answer protocol' so I am not able to read CAN data passively. 

Need your expertise input on the same.

Som4567, 2022-04-30:
It was wiring issues. finally resolved it. Thanks for all your support.

closed 2022-04-30 by Som4567