autowp/arduino-mcp2515 #82: How to send / Received extended Frame

sfJonas, 2022-06-17:
Hello,

I have two Arduino with a MCP2515 module on each. I installed your library and ran your CAN_write code on one, and your CAN_read code on the other.  I did these changes in the code, to send the extended frame :
 canMsg2.can_id  = 0x18DAF41F | CAN_EFF_FLAG;
mcp2515.sendMessage(MCP2515::TXB1, &canMsg2);
The sender transmits the message  with canMsg2.can_id  = 0x18DAF41F, But de read receives the message with canMsg2.can_id  = 98DAF41F
Could you help me to solve this problem?

maincraft-io, 2023-05-24:
Sender side:
```
packet.can_id = 0x12345678 | CAN_EFF_FLAG;
packet.can_dlc = 6;
```

Reciever side: 
```
canid_t can_id = canMsg.can_id;
can_id = can_id & (~CAN_EFF_FLAG);
```

autowp, 2025-02-26:
> Has same problem and fixed by this trick :)
> It's just demo how to CUT 0x8..... (**do not use it in another situations**)
> 
> Sender side:
> ```
> packet.can_id = 0x12345678 | CAN_EFF_FLAG;
> packet.can_dlc = 6;
> ```
> 
> Reciever side: 
> ```
> canid_t can_id = canMsg.can_id;
> can_id = can_id & (~CAN_EFF_FLAG);
> ```

Looks like the only right solution

closed 2025-02-26 by autowp