autowp/arduino-mcp2515 #88: Sending messages cannot be updated

hanlu1215, 2023-03-24:
In the send example, if each time the message to be sent is updated in a loop, it is found that the update cannot be completed, and the CAN bus is always the first third message

HarveyBates5B, 2023-03-26:
I'm having the same issue, I'm thinking that the the MCP_CANINTF register is not being reset resulting in all the message buffers not being able to be re-assigned.

HarveyBates5B, 2023-03-26:
This is the output log I get when trying to send a message:
```
Send msg - OK <- I can see on the oscilloscope it being sent
Send msg - OK <- I can see on the oscilloscope the data has been updated
Send msg - OK
Send msg - All TX Busy <- No change in data on the oscilloscope 
Send msg - All TX Busy
...
```

HarveyBates5B, 2023-03-26:
Ended up modifying the `sendMessage()` method to reset the TX CTRL registers if all the buffers are full. Don't know what the implications are so if someone who knows more about this could help that would be greatly appreciated.

This is my sendMessage() method now, be aware its recursive.
```c++
MCP2515::ERROR MCP2515::sendMessage(const struct can_frame *frame)
{

    if (frame->can_dlc > CAN_MAX_DLEN) {
        return ERROR_FAILTX;
    }

    TXBn txBuffers[N_TXBUFFERS] = {TXB0, TXB1, TXB2};

    for (auto & txBuffer : txBuffers) {
        const struct TXBn_REGS *txbuf = &TXB[txBuffer];
        uint8_t ctrlval = readRegister(txbuf->CTRL);
        if ( (ctrlval & TXB_TXREQ) == 0 ) {
            return sendMessage(txBuffer, frame);
        }
    }

    uint8_t zeros[14]{};
    setRegisters(MCP_TXB0CTRL, zeros, 14);
    setRegisters(MCP_TXB1CTRL, zeros, 14);
    setRegisters(MCP_TXB2CTRL, zeros, 14);

    return sendMessage(frame);
}
```

closed 2023-03-27 by hanlu1215