autowp/arduino-mcp2515 #90: Filter Example that Works

tonyc770, 2023-04-04:
I've come across a solution for utilizing the Masks and Filters in the MCP2515 Library. This was through the help of Dr. Dogan Ibahim as I have his book on CAN bus projects. It appears that all the Masks and Filters have to be configured or other CAN ID's that were not specifically filtered will pass though. Here is an example that works.

This example will only Filter on CAN ID 0x1AD. Notice that both Masks and all 6 filters have to be configured. I am setting both Masks to 0x7FF as to check each bit. You can adjust as needed to allow sequential ID's.

This is the snippet that would be in the Arduino setup() function.

`mcp2515.reset();
  mcp2515.setBitrate(CAN_250KBPS, MCP_8MHZ);
  mcp2515.setConfigMode();
  //Set Mask 0 to Check All Bits (0x7FF)
  mcp2515.setFilterMask(MCP2515::MASK0, false, 0x7FF);
  //Set Filters 0 and 1 to Filter Only CAN ID on 0x1AD
  mcp2515.setFilter(MCP2515::RXF0, false, 0x1AD);
  mcp2515.setFilter(MCP2515::RXF1, false, 0x1AD);
  //Set Mask 1 to Check All Bits (0x7FF)
  mcp2515.setFilterMask(MCP2515::MASK1, false, 0x7FF);
  //Set Filters 2 to 5 to Filter Only CAN ID on 0x1AD
  mcp2515.setFilter(MCP2515::RXF2, false, 0x1AD);
  mcp2515.setFilter(MCP2515::RXF3, false, 0x1AD);
  mcp2515.setFilter(MCP2515::RXF4, false, 0x1AD);
  mcp2515.setFilter(MCP2515::RXF5, false, 0x1AD);

  mcp2515.setNormalMode();
}`

To check other ID's simply add them to any of the 6 filters.

Hope this helps!!

-Tony

burger1473, 2023-04-06:
Hello, how is the code to read the FIFO message? Thank you

tonyc770, 2023-04-07:
Hi burger1473 - I don't understand your FIFO question...
The code snippet is just showing how to setup the masks and filters, to read the actual data refer to the code library examples.

-Tony

maincraft-io, 2023-04-08:
> I've come across a solution for utilizing the Masks and Filters in the MCP2515 Library. This was through the help of Dr. Dogan Ibahim as I have his book on CAN bus projects. It appears that all the [...]

Thank you. Your code sampe is help to me. :D

maincraft-io, 2023-05-24:
I will share also code snipped for extended id: 
```
mcp2515.reset();
mcp2515.setBitrate(CAN_250KBPS, MCP_8MHZ);
mcp2515.setConfigMode();

// Set both masks to check all 29 bits (0x1FFFFFFF)
mcp2515.setFilterMask(MCP2515::MASK0, true, 0x1FFFFFFF);
mcp2515.setFilterMask(MCP2515::MASK1, true, 0x1FFFFFFF);

// Set all filters to only allow CAN ID 0x1A1010B1
mcp2515.setFilter(MCP2515::RXF0, true, 0x1A1010B1);
mcp2515.setFilter(MCP2515::RXF1, true, 0x1A1010B1);
mcp2515.setFilter(MCP2515::RXF2, true, 0x1A1010B1);
mcp2515.setFilter(MCP2515::RXF3, true, 0x1A1010B1);
mcp2515.setFilter(MCP2515::RXF4, true, 0x1A1010B1);
mcp2515.setFilter(MCP2515::RXF5, true, 0x1A1010B1);

mcp2515.setNormalMode();
```