autowp/arduino-mcp2515 #97: Sends or Receives Message 1 or Message 2 but not both.

parentevan, 2023-05-31:
Hey there. I'm able to send or maybe receive message 1 or 2 if I comment out a send line, but I am NOT able to receive both messages.
I'm running Arduino Nanos with the standard cheap MCP2515 boards with the terminating resistors shunted. Any ideas how I may be able to remedy this? 

16:13:45.325 -> F6 8 8E 87 32 FA 26 8E BE 86 
16:13:45.436 -> F6 8 8E 87 32 FA 26 8E BE 86 
16:13:45.528 -> F6 8 8E 87 32 FA 26 8E BE 86 
16:13:45.637 -> F6 8 8E 87 32 FA 26 8E BE 86

maincraft-io, 2023-06-01:
Your serial port doesn't print correctly results.

parentevan, 2023-06-01:
Very odd. Today it works fine. Both messages sending and receiving. I appreciate the feedback!

maincraft-io, 2023-06-02:
I has same issues when i'm try to use serial print to display results.
Use some buffer for collect your data and later print/display it. 
Also, use some delays between multiple sending data (1-2ms)