autowp/arduino-mcp2515 #99: Interrupts work partially

dimitris12kara, 2023-06-28:
When I use the attachInterrupt(digitalPinToInterrupt(2),function,FALLING)
it works only in some setups I use. What parameters could affect my project? I think something has to do with terminal resistors.
In the projects that the interrupt function doesn't work, I can still read the messages if I put the same function inside the loop.The interrupt pin in these situations falls only for the first message but then it stays zero, no matter if it receives messages or not.
Any help would be appreciated!

ISbitPro, 2025-04-15:
I also have propblem with the interrupt pin. It feels like it doesnt always go high again
I check a receive flag instead of interrupt pin now, and that always works for me.

`if(CAN0.checkReceive() == CAN_MSGAVAIL){`
instead of 
`if(!digitalRead(CAN1_INT)) { `

MatiasJHZ, 2025-05-19:
I am having the same problem. Do you use any pull up resistor in the INT pin of the MCP2515?

dimitris12kara, 2025-05-19:
> I am having the same problem. Do you use any pull up resistor in the INT pin of the MCP2515?

It doesn't need a pull-up.  I solved the problem by setting the bits CANINTF_RX0IF and CANINTF_RX1IF in the MCP2515_CANINTF register after each time I received a message. Maybe even add it to the setup function.

it must be something like this:
mcp2515.setRegister(MCP_CANINTF, CANINTF_RX0IF | CANINTF_RX1IF)